linorobot/linorobot #1: Wheels

burf2000, 2017-02-13:
Hi there, do you still do ROS?  I am building a similar robot and I am stuck on 1 thing which I would absolutely love help with.

I am using Hector Slam which produces /odom etc for my lidar.  I have a URDF.  How do I link the wheel in the mode (mine is lwheel, yours is wheel_left) to the system as it always shows as an error in RVIZ?

grassjelly, 2017-02-13:
hi @burf2000 You can start a thread here https://groups.google.com/forum/#!forum/linorobot so the rest can see your question as well.

burf2000, 2017-02-14:
Done, if someone can help me, I can pay in BEER, love or whatever!

grassjelly, 2017-02-28:
Closing this @burf2000 mate! Thanks!

closed 2017-02-28 by grassjelly

grassjelly, 2018-03-20:
Posting the opened thread here: https://groups.google.com/forum/#!searchin/linorobot/hector%7Csort:date/linorobot/v3s6olorGIo/crnkGd7uCgAJ