linorobot/linorobot #3: Intel RealSense R200 support

jabrena, 2017-07-15:
I read that you are thinking to use Intel RealSense R200 in your project. I thought that that depth camera is not possible to be installed. Do you know some way to run ln raspi3?
https://github.com/linorobot/linorobot/wiki/1.-Getting-Started

grassjelly, 2017-07-15:
I haven't really tried the R200 on an arm yet. No leads on how to make it work on Pi3 but I've seen this guy running on a Tx1 https://github.com/IntelRealSense/librealsense/issues/73#issuecomment-198649000 . Nonetheless, I'll still release it - for those who are using x86 laptops for their builds. ;)

jabrena, 2017-07-15:
Ok.

Many thanks.

closed 2017-07-15 by jabrena

grassjelly, 2017-07-31:
Just to follow up on this, I found this repository to install librealsense on Jetson boards. Not sure if this will work on a Pi, but might be worth the try. https://github.com/jetsonhacks/installLibrealsense/blob/master/arm.patch
https://www.youtube.com/watch?v=yGhfoxReUf4