linorobot/linorobot #4: Doubt about /tf

jabrena, 2017-07-17:
Hi @grassjelly,

I write you because I have a doubt about your robots.
In what part do you define the relation between the base_frame with the laser_frame?
How to publish the tf topic and the transformation?

Maybe the question are a bit basic.

Many thanks in advance.

Juan Antonio

grassjelly, 2017-07-17:
You can define your laser's transform here: https://github.com/linorobot/linorobot/blob/master/launch/laser.launch#L6

It's also written in the Wiki. :P I recommend to please post questions on the forum and only post code related issues here. Thanks! https://groups.google.com/forum/#!forum/linorobot

grassjelly, 2017-07-20:
Closing this down. Feel free to post in the forum for further questions. Cheers!

closed 2017-07-20 by grassjelly