linorobot/linorobot #6: PID calibration for 4wd

aaravrav142, 2017-10-17:
Hi,

I have finished making a heavy 4wd robot following the guide. The robot currently weighs over 20 kgs with 10 inch wheels. The build and installation of the code went flawless, no problems whatsoever and I thank you for this great opensource code. I am however clueless as to how to configure the PID values. The robot can run straight forward and backward but not able to turn at its place. I am not sure how should I start determining the best value for k_p, k_d and k_i.  I also followed the guide that you have included but it was not helpful. Would be happy to get some feedback. Thanks

grassjelly, 2017-10-17:
Hi,

Can you post to the group https://groups.google.com/forum/#!forum/linorobot instead? 

Thanks.

grassjelly, 2017-10-17:
Closing this since it’s not code related. Here’s the groups thread for reference: https://groups.google.com/forum/m/?utm_medium=email&utm_source=footer#!msg/linorobot/PfGbdzls2Yo/TFEtOQ7-AAAJ

closed 2017-10-17 by grassjelly