linorobot/linorobot #10: Compile Problems - Teensy 3.5/3.6

MORP, 2018-03-10:
When trying to compile and flash the Teensy firmware a number of errors pop-up when using Teensy 3.5 or 3.6 as target platforms. The reason is that the file "direct_pin_read.h", located in ... /src/linorobot/teensy/firmware/lib/encoder/utility only addresses Teensy 3.1 and 3.2 platforms. An easy fix is to replace line 4 by

#if defined(__AVR__) || defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__)

Cheers

 Chris

grassjelly, 2018-03-10:
Thanks for reporting the issue. Patched it and will include in the next merge (New IMU support). https://github.com/linorobot/linorobot/commit/5191cb9adc340b624c64b5fbae7f09287f4bfb0b

I'll leave this open for now.

grassjelly, 2018-03-17:
Patch has been merged to master. Closing this now.

closed 2018-03-17 by grassjelly