linorobot/linorobot #22: The ESC  can not work well  because the pid error too large

zhangcaocao, 2019-01-17:
I have a Rc car , but when i make my teensy Output pwm wave ，my esc can not work, my motor is not runing. The i find the reason of it is the pid error too large!!

so I add blow code to make the pid error is not too large ! (PID.cpp)
`double PID::compute(float setpoint, float measured_value)
{
    double error;
    double pid;

    //setpoint is constrained between min and max to prevent pid from having too much error
    error = setpoint - measured_value;

    if (error > 0)
    {
        if (error >= max_error)
        {
            error = max_error;
        }
    }else{
        if (error <= -max_error)
        {
            error = -max_error;
        }

    }

    integral_ += error;
    derivative_ = error - prev_error_;

    if(setpoint == 0 && error == 0)
    {
        integral_ = 0;
    }

    pid = (kp_ * error) + (ki_ * integral_) + (kd_ * derivative_);
    prev_error_ = error;

    return constrain(pid, min_val_, max_val_);
}`

AllofLife, 2019-05-12:
你需要自己标定pid

gpmontt, 2019-05-12:
Hi everyone, @zhangcaocao have you tried to calibrate your PID? normally PWM value work beetween value between 1ms and 2ms  as a [servo](https://ebldc.com/?p=48)

maybe, you need to give the correct values in your PID instance. 
 
```arduino
#include "PID.h"
PID pid( -255, 255, 0.05, 0.9, 0.1);
/*PID(float min_val, float max_val, float kp, float ki, float kd)
 * min_val = min output PID value
 * max_val = max output PID value
 * kp = PID - P constant
 * ki = PID - I constant
 * di = PID - D constant
 */

float setpoint = 30;
float measured_value = 0;
void setup() 
{
    Serial.begin(9600);
}

void loop() 
{
    Serial.println(pid.compute(setpoint, measured_value));
    delay(1000);
    measured_value++;
}
```

zhangcaocao, 2019-05-24:
@AllofLife ，不是标定问题好吧，而是PID的error太大了，输入到电调中电机无法正常动作。

closed 2019-07-27 by grassjelly

zhuwei, 2019-12-16:
我觉得跟pid没关系啊。
源码直接把rpm写入pwm了。
问题是pwm是8位，最大也就255，但rpm不一定，有可能600。
我觉得应该按比例转换成pwm，而不是把rpm当做pwm。
搞不清楚作者为什么这么做。

zhuwei, 2019-12-16:
@zhangcaocao @AllofLife 大佬，能不能加个联系方式。PID整定不会，能不能指导一下。

AllofLife, 2019-12-16:
wechat r15297833965