linorobot/linorobot #31: ROS Melodic (Ubuntu 18)

silvermind20, 2019-07-04:
Hi,
Would this work on ROS Melodic (Ubuntu 18)?

grassjelly, 2019-07-05:
The hardware drive shouldn’t have any problem. The installer would throw an error once it hit gmapping package though as it’s not released in Melodic yet. Alternatively, you can comment it out from the installer and build it from source.

grassjelly, 2019-07-27:
Working Melodic installer here:
https://github.com/linorobot/lino_install/tree/melodic
https://travis-ci.org/linorobot/lino_install/builds/564405465

Few caveats:
1. Kinect Driver's not working
2. python-gudev 's not working

sdey76, 2019-07-28:
Hi
On BTS7960 Motor controller two enable pins. 
Do we need to externally give 5 volt. As the wiring has pwm ,ina inb pins in diagram.

grassjelly, 2019-07-31:
@sdey76 Hi, please post your questions here: 
https://groups.google.com/forum/m/#!forum/linorobot .

karthikbalu, 2019-08-06:
I can confirm that we managed to run linorobot melodic on Jetson Nano with few glitches with udev rules and solved it

arminf82, 2019-09-06:
> Working Melodic installer here:
> https://github.com/linorobot/lino_install/tree/melodic
> https://travis-ci.org/linorobot/lino_install/builds/564405465
> 
> Few caveats:
> 
>     1. Kinect Driver's not working
> 
>     2. python-gudev 's not working

Will everything work when I simply install the Kinect drivers on my own?

grassjelly, 2019-09-06:
@arminf82  Yes you can compile it from source

TCIII, 2019-09-25:
@grassjelly,

I am running Ubuntu Mate 18.0.4 and the ROS Melodic-base on a Rpi 3B+.

The gmapping package is available for ROS Melodic now and I have successfully installed it.

Your  https://github.com/linorobot/lino_install/tree/melodic link is broken as I get a "does not exist" message when running sudo git clone https://github.com/linorobot/lino_install/tree/melodic or am I doing something wrong?

I assume that I can run git clone https://github.com/linorobot/lino_install and then modify the install file before I run ./install base sensor? 

According to Python bindings for gudev, gudev was removed 12/16/2017. The source is available, but I need to learn how to build and install gudev on my own.

Should I be posting on your Google forum?

grassjelly, 2019-09-25:
Hi you should change the branch to melodic and not directly clone the link:

git clone https://github.com/linorobot/lino_install
cd lino_install
git checkout melodic

Once done, you can run the installer.

Hope this helps.

TCIII, 2019-09-25:
@grassjelly,

Thanks for the suggestion, much appreciated.

Any suggestions concerning installing python-gudev on Ubuntu Mate 18.0.4? I am new at Python/Linux Programming and would appreciate any help.

TCIII, 2019-09-26:
Hi All,

I have attempted to install Linorobot on Ubuntu Mate/ROS Melodic on my Rpi 3B+ using the Melodic branch of lino_install.

The install runs fine until it gets to "[ 46%] Building CXX object lino_pid/CMakeFiles/pid_listen.dir/src/pid_listen.cpp.o" at which time this message appeared "c++: internal compiler error: Killed (program cc1plus)".

The build then continued to the [50%] point at which time this message appeared and the program went to the command prompt:

Makefile:140: recipe for target 'all' failed

make: *** [all] Error 2

Invoking "make -j4 -l4" failed 

I am attaching a file of the failure information from [46%] to the command line prompt.

Regards,
TCIII

[Linorobot Install Failure Message2_20190926.txt.txt](https://github.com/linorobot/linorobot/files/3659502/Linorobot.Install.Failure.Message2_20190926.txt.txt)

TCIII, 2019-09-26:
@grassjelly,

I would say that the Ubuntu Mate/ROS Melodic combination might not be doable at the moment.
I say this because even though the ROS gmapping package is available for Melodic, there presently is no source for building and installing python-gudev for ROS Melodic.

I therefore would appreciate it if you could direct me to a source for a bootable image of Ubuntu 16.04 for the Rpi 3B+ if at all possible as I have not been able to find one that will work.

Regards,
TCIII

imeshsps, 2019-12-11:
@TCIII python-gudev is used to assign names to serial ports right ? I commented it out from installer file and used [this](https://askubuntu.com/questions/905115/assigning-name-to-usb-device/906206) guide to add names manually. So far i got ROS serial working fine. Hope this helps someone.

Geekgineer, 2020-02-07:
#### The solution for udev rules is to build python-gudev from source
```sh
$ git clone https://github.com/nzjrs/python-gudev.git
$ cd python-gudev
$ sudo apt install libtool-bin
$ sudo apt install python-gobject-2-dev
$ ./autogen.sh 
$ make
$ sudo make install
```
This will install the python-gudev to your sytem so **lino_udev.py** can  ``` import gudev```
_tested on Ubuntu 18.04.3_

Valentinkvn, 2020-02-09:
@Geekgineer,

Hi, I also use Ubuntu 18.04.3 installed on a Nvidia Jetson Nano.
I followed your set of instructions but at the `./autogen.sh` command, I received the following error.

```
checking for GUDEV... no
configure: error: Package requirements (gudev-1.0 >= 210) were not met:

No package 'gudev-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables GUDEV_CFLAGS
and GUDEV_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

quicktwit, 2020-03-21:
> @Geekgineer,
>
> Hi, I also use Ubuntu 18.04.3 installed on a Nvidia Jetson Nano.
> I followed your set of instructions but at the `./autogen.sh` command, I received the following error.
>
> ```
> checking [...]
Try this:
`sudo apt-get install libgudev-1.0-dev`

chopsuey5000, 2020-06-26:
Just in case, I've installed Linorobot in Raspberry Pi 3B with Ubuntu Mate 18.04. These are the steps i made:

**Install ubuntu mate 18.04** - https://ubuntu-mate.org/download/armhf/

After the installation:
```
sudo apt update
sudo apt install git
sudo apt install python-pip
```

**Install ros melodic**
```
git clone https://github.com/linorobot/rosme
cd rosme
./install
source /opt/ros/melodic/setup.bash
```

**Create a ROS workspace** - http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment
```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/
catkin_make
source devel/setup.bash
```

**Linorobot Installation**
```
git clone https://github.com/linorobot/lino_install
cd lino_install
git checkout melodic
```

uDev Rules
```
sudo apt install python-gobject
sudo apt install libtool-bin
sudo apt install python-gobject-2-dev
sudo apt install autoconf
sudo apt-get install libgudev-1.0-dev
git clone https://github.com/nzjrs/python-gudev.git
cd python-gudev
./autogen.sh 
make
sudo make install

rosrun lino_udev lino_udev.py
sudo cp 58-lino.rules /etc/udev/rules.d/58-lino.rules
```
Restart udev:
```
sudo service udev reload
sudo service udev restart
```
Confirm if the uDev rules worked:
```
ls /dev/linobase
ls /dev/linolidar
```

If the serial ports cannot be detected, restart the robot's computer and check again. Take note that freshly shipped Teensy boards has to be be programmed once before you can see its ports.

In my case i had to restart to be able to list the ports.

ngadt12, 2020-11-29:
git clone https://github.com/linorobot/lino_install
cd lino_install
git checkout melodic
then: 
cd lino_install
catkin_make
cd ~/linorobot_ws/
./install 2wd rplidar
source devel/setup.bash

saisd138, 2023-04-17:
can anyone help me out im stuck here I've been getting this error I've followed [this](https://github.com/linorobot/linorobot/issues/31#issuecomment-650132815) method and im facing this error

ubuntu@ubuntu:~$ rosrun lino_udev lino_udev.py

[rospack] Error: package 'lino_udev' not found