linorobot/linorobot #32: Ros 2 migration

karthikbalu, 2019-07-07:
Any plan for ROS 2 Migration

I guess it’s not too complicated, pls correct me 

Thanks

grassjelly, 2019-07-07:
Hi,

You can check out the initial efforts here. https://github.com/linorobot2/linorobot2.

karthikbalu, 2019-07-07:
Is it possible to test the migration at this stage now

Thanks & awesome work👍👍👍👍👏👏

grassjelly, 2019-07-07:
Sure, feel free to give it a try. :)

There's a few caveats though:
1. It's been a year and it wasn't properly documented as it was totally an experiment how the stack can be ported to ROS2.
2. The whole architecture might change in the future as this approach is using a stm32 uC. I'm planning to re-use the same hardware design as possible to make it easier for old users to port their systems to ROS2.

Having said so, I foresee linorobot2 to still run on Teensy with a ROS2 Arduino library running as the backbone infrastructure.

karthikbalu, 2019-07-07:
😎awesome I will give it a try, any reason stm32 was chosen instead of teensy, I thought I could give it a try with existing  teensy as both are Arduino compatible with some code changes if needed

Btw I have an extensive troubleshooting documentation for Linorobot with all the gaps filled while I built the robot,
 where can I post it? A new file with a pull request?

grassjelly, 2019-07-09:
Why STM32? Because the initial efforts on running XRCE w/ NuttX on a uC was implemented in STM32. 

Replied your PR. Closing this for now :)

closed 2019-07-09 by grassjelly