linorobot/linorobot #41: Scanse Sweep compatibility

Valentinkvn, 2020-04-24:
Hi,

I'm using linorobot with Jetson Nano (Ubuntu 18.04), Arduino Mega and Sweep Lidar.
I configured the linobase module and the minimal.launch is working fine.

In order to be able to use Sweep Lidar I added it's package into the linorobot directory and catkin_make it. After that, I changed the linorobot launch files to run the sweep lidar launch file. 
When I launch the bringup.launch file, I receive those error.

My question is: What else would need to be done in order to use the Sweep Lidar?

```
started roslaunch server http://192.168.43.50:34635/

SUMMARY
========

PARAMETERS
 * /apply_calib/calib_file: /home/jetson/lino...
 * /apply_calib/calibrate_gyros: True
 * /ekf_localization/base_link_frame: base_footprint
 * /ekf_localization/diagnostics_agg: True
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: /imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_relative: True
 * /ekf_localization/odom0: /raw_odom
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: True
 * /ekf_localization/odom0_relative: False
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /imu_filter_madgwick/fixed_frame: base_footprint
 * /imu_filter_madgwick/orientation_stddev: 0.05
 * /imu_filter_madgwick/publish_tf: False
 * /imu_filter_madgwick/use_mag: True
 * /imu_filter_madgwick/use_magnetic_field_msg: True
 * /imu_filter_madgwick/world_frame: enu
 * /pointcloud_to_laserscan/angle_increment: 0.0174533
 * /pointcloud_to_laserscan/angle_max: 3.14
 * /pointcloud_to_laserscan/angle_min: -3.14
 * /pointcloud_to_laserscan/concurrency_level: 1
 * /pointcloud_to_laserscan/max_height: 1.0
 * /pointcloud_to_laserscan/min_height: -1.0
 * /pointcloud_to_laserscan/range_max: 40.0
 * /pointcloud_to_laserscan/range_min: 0.0
 * /pointcloud_to_laserscan/scan_time: 0.1
 * /pointcloud_to_laserscan/target_frame: laser
 * /pointcloud_to_laserscan/transform_tolerance: 0.001
 * /pointcloud_to_laserscan/use_inf: True
 * /rosdistro: melodic
 * /rosserial_lino/baud: 57600
 * /rosserial_lino/port: /dev/linobase
 * /rosversion: 1.14.5
 * /sweep_node/frame_id: laser
 * /sweep_node/serial_baudrate: 115200
 * /sweep_node/serial_port: /dev/linolidar

NODES
  /
    apply_calib (imu_calib/apply_calib)
    base_footprint_to_base_link (tf2_ros/static_transform_publisher)
    base_footprint_to_imu_link (tf2_ros/static_transform_publisher)
    base_link_to_laser (tf2_ros/static_transform_publisher)
    ekf_localization (robot_localization/ekf_localization_node)
    imu_filter_madgwick (imu_filter_madgwick/imu_filter_node)
    lino_base_node (linorobot/lino_base_node)
    pointcloud_to_laserscan (pointcloud_to_laserscan/pointcloud_to_laserscan_node)
    rosserial_lino (rosserial_python/serial_node.py)
    sweep_node (sweep_ros/sweep_node)

auto-starting new master
process[master]: started with pid [26933]
ROS_MASTER_URI=http://192.168.43.50:11311

setting /run_id to 11079158-8627-11ea-ad52-12ed0dd4ce4d
process[rosout-1]: started with pid [26944]
started core service [/rosout]
process[rosserial_lino-2]: started with pid [26951]
process[apply_calib-3]: started with pid [26952]
process[imu_filter_madgwick-4]: started with pid [26953]
process[base_footprint_to_imu_link-5]: started with pid [26954]
[ INFO] [1587731298.663476387]: Starting ImuFilter
[ INFO] [1587731298.677961559]: Using dt computed from message headers
[ INFO] [1587731298.724266254]: Imu filter gain set to 0.100000
[ INFO] [1587731298.724941426]: Gyro drift bias set to 0.000000
[ INFO] [1587731298.725399145]: Magnetometer bias values: 0.000000 0.000000 0.000000
process[lino_base_node-6]: started with pid [26965]
process[base_footprint_to_base_link-7]: started with pid [26971]
process[ekf_localization-8]: started with pid [26972]
process[sweep_node-9]: started with pid [26983]
process[pointcloud_to_laserscan-10]: started with pid [26985]
process[base_link_to_laser-11]: started with pid [26987]
[ WARN] [1587731299.346980673]: Both imu0_differential and imu0_relative were set to true. Using differential mode.
[INFO] [1587731300.024167]: ROS Serial Python Node
[INFO] [1587731300.042407]: Connecting to /dev/linobase at 57600 baud
[INFO] [1587731302.160059]: Requesting topics...
Error: invalid response header checksum
[INFO] [1587731302.361959]: Note: publish buffer size is 512 bytes
[INFO] [1587731302.367370]: Setup publisher on raw_imu [lino_msgs/Imu]
[INFO] [1587731302.380240]: Note: subscribe buffer size is 512 bytes
[INFO] [1587731302.384288]: Setup subscriber on pid [lino_msgs/PID]
[INFO] [1587731302.396869]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1587731302.403159]: LINOBASE CONNECTED
[ERROR] [1587731302.409229]: Tried to publish before configured, topic id 125
[INFO] [1587731302.413351]: Requesting topics...
[ERROR] [1587731302.429750]: Tried to publish before configured, topic id 125
[INFO] [1587731302.433974]: Requesting topics...
[sweep_node-9] process has finished cleanly
log file: /home/jetson/.ros/log/11079158-8627-11ea-ad52-12ed0dd4ce4d/sweep_node-9*.log
[INFO] [1587731302.457929]: Setup publisher on raw_imu [lino_msgs/Imu]
[ERROR] [1587731302.488946]: Tried to publish before configured, topic id 125
[INFO] [1587731302.493745]: Requesting topics...
[INFO] [1587731302.538571]: Setup publisher on raw_vel [lino_msgs/Velocities]
[INFO] [1587731302.549725]: Setup publisher on raw_imu [lino_msgs/Imu]
[INFO] [1587731302.622646]: Setup publisher on raw_vel [lino_msgs/Velocities]
[INFO] [1587731302.638786]: Setup publisher on raw_imu [lino_msgs/Imu]
[ INFO] [1587731302.693795733]: Calibrating gyros; do not move the IMU
[ WARN] [1587731308.828438073]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[ INFO] [1587731310.355790401]: Gyro calibration complete! (bias = [-0.056, 0.037, -0.012])
[ INFO] [1587731310.508411130]: First pair of IMU and magnetometer messages received.

```

grassjelly, 2020-04-27:
Hi,
Can you share your launch files please? Thanks

Valentinkvn, 2020-05-04:
Hi,

Sorry for late response. I configured the environment as you said but the error still occurs.
These are the launch files.

**../linorobot/launch/include/laser.launch**
```
<launch>
    <!-- Run Linorobot compatible laser drivers. Takes reference from env var LINOLIDAR. ie. export LINOLIDAR=xv11 -->
    <include file="$(find linorobot)/launch/include/lidar/sweep.launch" />
    <!-- Publish static transform of the laser. Define your sensor offset here -->
    <node pkg="tf2_ros" type="static_transform_publisher" name="base_link_to_laser" args="0.065 0 0.098 0 0 0  /base_link /laser"/>
</launch>
```

**../linorobot/launch/include/lidar/sweep.launch** (here I added the conversion between pc2 and laserscan)
```
<launch>
    <!-- run sweep_node node -->
    <node name="sweep_node"          pkg="sweep_ros"  type="sweep_node" output="screen">
            <param name="serial_port"         type="string" value="/dev/linolidar"/>
            <param name="serial_baudrate"     type="int"    value="115200"/>
            <param name="frame_id"            type="string" value="laser"/>
    </node>

    <!-- run pointcloud_to_laserscan node -->
    <node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">

        <remap from="cloud_in" to="pc2"/>
        <rosparam>
            target_frame: laser # Leave disabled to output scan in pointcloud frame
            transform_tolerance: 0.001
            min_height: -1.0
            max_height: 1.0

            angle_min: -3.14 # -M_PI/2
            angle_max: 3.14 # M_PI/2
            angle_increment: 0.0174533 # M_PI/360.0
            scan_time: 0.1
            range_min: 0.0
            range_max: 40.0
            use_inf: true

            # Concurrency level, affects number of pointclouds queued for processing and number of threads used
            # 0 : Detect number of cores
            # 1 : Single threaded
            # 2->inf : Parallelism level
            concurrency_level: 1
        </rosparam>
    </node>
</launch>

```

grassjelly, 2020-05-05:
Your launch files looks good. Just omit the pointcloud_to_laserscan. 

Also make sure that sweep publishes that data in "laser" frame. Otherwise, you have to rename "laser" in static_transform_publisher to the correct frame the LIDAR is using.

Valentinkvn, 2020-05-20:
The problem was not the lidar itself, it worked well, but the udev rules that i've manually created. I observed that each time when I plugged in a new lino device (LiDAR / Arduino Mega) each device was bound on the same ttyUSB* port.

So, I followed the https://github.com/linorobot/linorobot/issues/31#issuecomment-602075774 instructions to configure the libgudev on my ubuntu 18.04, and after the port configurations provided by the lino_udev script, I still had to manually configure the port for Arduino Mega. 

But it works now, thank you!

closed 2020-05-20 by Valentinkvn