linorobot/linorobot #44: From teenzy to Arduino Mega2560-tutorial

zoldaten, 2020-07-09:
Just want to post a tutorial how to use Arduino Mega2560 instead Teenzy 3.1/3.2.
This tutorial is my own collection of headache while compiling code with platformio )
So everything works for date 07.2020 by this tutorial.

1. find your board first with udev:
**rosrun lino_udev lino_udev.py**
    connect arduino
    save udev
**sudo cp 58-lino.rules /etc/udev/rules.d/58-lino.rules
reboot**

2.Change settings in platformio.ini
**cd /linorobot_ws/src/linorobot/teensy/firmware
nano platformio.ini**
    [env:teensy31]
    platform = teensy
    framework = arduino
    board = teensy31

    ;[env:mega]
    ;platform = atmelavr
    ;framework = arduino
    ;board = megaatmega2560

    upload_protocol = teensy-cli
    upload_port = /dev/linobase
**to**
    ;[env:teensy31]
    ;platform = teensy
    ;framework = arduino
    ;board = teensy31

    [env:mega]
    platform = atmelavr
    framework = arduino
    board = megaatmega2560

    upload_protocol = avrispmkII
    upload_port = /dev/linobase

3.install missed library 
**platformio lib -g install Servo**
or
Put the servo library in the lib folder
linorobot/teensy/firmware/lib/PUTHERE
and change firmware.ino
#include <Servo.h> --> #include "Servo.h"

4.comment 26 line in firmware.ino
cd /linorobot_ws/src/linorobot/teensy/firmware/src
nano firmware.ino
5.upload firmware:
cd /linorobot_ws/src/linorobot/teensy/firmware
platformio run --target upload

zoldaten, 2020-07-09:
Just want to post a tutorial how to use Arduino Mega2560 instead Teenzy 3.1/3.2.
This tutorial is my own collection of headache while compiling code with platformio )
So everything works for date 07.2020 by this tutorial.

1. find your board first with udev:
**rosrun lino_udev lino_udev.py**
    connect arduino
    save udev
**sudo cp 58-lino.rules /etc/udev/rules.d/58-lino.rules
reboot**

2.Change settings in platformio.ini
**cd /linorobot_ws/src/linorobot/teensy/firmware
nano platformio.ini**
    [env:teensy31]
    platform = teensy
    framework = arduino
    board = teensy31

    ;[env:mega]
    ;platform = atmelavr
    ;framework = arduino
    ;board = megaatmega2560

    upload_protocol = teensy-cli
    upload_port = /dev/linobase
**to**
    ;[env:teensy31]
    ;platform = teensy
    ;framework = arduino
    ;board = teensy31

    [env:mega]
    platform = atmelavr
    framework = arduino
    board = megaatmega2560

    upload_protocol = avrispmkII
    upload_port = /dev/linobase

3.install missed library 
**platformio lib -g install Servo**
or
Put the servo library in the lib folder
linorobot/teensy/firmware/lib/PUTHERE
and change firmware.ino
#include <Servo.h> --> #include "Servo.h"

4.comment 26 line in firmware.ino
cd /linorobot_ws/src/linorobot/teensy/firmware/src
nano firmware.ino
5.upload firmware:
cd /linorobot_ws/src/linorobot/teensy/firmware
platformio run --target upload

closed 2020-07-09 by zoldaten

Snyp54, 2020-10-21:
Hi, can you have a solution for ftdi plug into mega 2560 pro mini?
When i run  rosrun lino_udev lino_udev.py ftdi & lidar detected but when i reboot after  sudo cp 58-lino.rules /etc/udev/rules.d/58-lino.rules, the ftdi it's nor recognized.