linorobot/linorobot #50: Use of integers with getRPM causes issues with robots with large wheels.

madgrizzle, 2020-09-29:
The robot I built uses a 10-inch wheel (with a 12-inch wheel base) and I was having problems with getting the robot to achieve an accurate orientation.  With a goal tolerance of 5 degrees, I was getting about +/-10 degrees.  While digging into the problem, I discovered that because of the size of the wheel, the RPM required to home in on an orientation of about less than 5 degrees would require an RPM of less than 1.  But because getRPM() in encoder.h returns an integer, a 0.999 RPM gets converted to 0 and the motors stop turning, even though the controller is still receiving twist commands.  I changed the code to return a float and it seemed to solve my issue.  Is there any reason it needs to be an integer?

grassjelly, 2020-09-29:
Hi,

Thanks for highlighting this error. The reason is that I took precedence from most motor RPM specs which are usually expressed in integer and with the assumption that anything in between a single RPM value is negligible (ie For control, the PID will handle and velocity calculation wise, EKF can do the corrections). As with your specs,  a hypothetical motor spinning at 1 RPM (12in wheel) will result a ~0.01 m/s linear velocity. However, I do agree casting this to float increases the resolution for such use cases as yours. 

Are you able to send a PR? 

Btw, can you confirm that when the returned RPM is 0 , the wheel stopped spinning. I think the behaviour should be the opposite, as the PID should try to spin the motor since there's a twist command given and the robot thinks that the wheel is not moving. I'll take a look as this might be  a potential bug. Thank you!

madgrizzle, 2020-09-29:
Yeah, I agree that you don't need much precision with RPM and it worked well enough to move around.. its just trying to get that high level of precision for orientation (wheels moving in opposite direction) that the float is needed. 

I'm not entirely sure I can do a PR with what I have because I've made a lot of other changes to the software to accommodate my specific setup, but I'll take a look (not a git expert).  But in general, just replaced int with float.

> Btw, can you confirm that when the returned RPM is 0 , the wheel stopped spinning. I think the behaviour should be the opposite, as the PID should try to spin the motor since there's a twist command [...]

Oh, sorry, it was late last night and it slipped my mind this morning.  A change is also needed to Kinematics because the structure that represents rpm also uses integers.  So when the kinematics converts twist commands to motor RPMs, it converts RPMs less than 1 to 0.  I checked and the only files I changed last night were kinematics.cpp, kinematics.h, and encoder.h.

grassjelly, 2020-10-05:
Thanks @madgrizzle  I'll include this in the next update.

closed 2020-10-05 by grassjelly