linorobot/linorobot #52: How use TEENSY 4.1 with linorobot

JamesFlosse, 2020-11-03:
Hi, i plan to use teensy 4.1 on your linorobot project.
Can you give me the processus to use it?
Best regards

Russ76, 2021-01-31:
James, the Teensy 4.1 will probably work fine, just be sure to consult a spec chart about the connections to the board, and go by name, not by position number on the board. I'm having trouble now using my Teensy 3.2 because Platformio finds the older Python version loaded earlier in installation, and refuses to run.

Russ76, 2021-02-01:
https://www.pjrc.com/store/teensy41.html
This site has lots of info about it. It is a lot more powerful than the 3.2 but still has PMW output pins. Some of Juan's code for the Teensy is elaborate (machine language) and may not work in a chip so different.

suryachereddy, 2021-06-01:
Did anyone test Teensy 4.1?

casakontar, 2021-06-27:
> Did anyone test Teensy 4.1?

I did and it was a nightmare. I adapted the libraries amd completely gave up on platformio. I used arduino ide.

JamesFlosse, 2022-01-30:
@casakontar can you please share your modify version please?