linorobot/linorobot #55: Intel RealSense R200 not available

Russ76, 2021-01-02:
Will the code work with the Intel D435 depth camera? The other is out-of-date.

grassjelly, 2021-01-19:
Hey Russ, 

Although not supported, D435 should work as long as topics matches the current real sense launch files.

Russ76, 2021-01-30:
I've got most of it installed but get an error with Platformio.
~/linorobot_ws/src/linorobot/teensy/firmware$ platformio ci --project-conf=./platformio.ini --lib="./lib/ros_lib" --lib="./lib/config"  --lib="./lib/motor"  --lib="./lib/kinematics"  --lib="./lib/pid"  --lib="./lib/imu" --lib="./lib/encoder"
Error: Python 3.6 or later is required for this operation. 
Please check a migration guide:
https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

Is there a legacy version I can use? Or should I go to Python 3.6?
Thanks, Russ

Russ76, 2021-01-30:
Got it mostly installed but this error comes up:

~/linorobot_ws/src/linorobot/teensy/firmware$ platformio ci --project-conf=./platformio.ini --lib="./lib/ros_lib" --lib="./lib/config"  --lib="./lib/motor"  --lib="./lib/kinematics"  --lib="./lib/pid"  --lib="./lib/imu" --lib="./lib/encoder"
Error: Python 3.6 or later is required for this operation. 
Please check a migration guide:
https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

I'm using Raspberry Pi 4 with 4 Gig RAM, Raspbian Buster and ROS Melodic. Is there a way to program Teensy with the Arduino IDE? I see your libraries needed for Teensy code are extensive. 

I installed Python 3.6 but still system reports 2.7 to Platformio and it refuses to run.
Russ