linorobot/linorobot #56: E: Unable to locate package python-gudev

pitosalas, 2021-01-12:
I am installing on Ubuntu 20.04. I've successfully installed ROS noetic. When attempting:

`./install 2wd ydlidar
`
I see this:

```
INSTALLING NOW....

Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
E: Unable to locate package python-gudev
```

I know that I am not the only one to see this but lots of googling hasn't solved it for me yet. So I have some questions:

1) Does the Linorobot software package depend on/support just python2, or python3, or either one?
2) Is there a known incompatibility with Ubuntu 20.04?
3) Is there a known incompatibility with ROS Noetic?

Thanks

grassjelly, 2021-01-19:
Hi Pito,

1. All python scripts were developed on python2. I haven't tested on 3 yet but should be pretty straight forward to fix backward incompatibilities.
2. I haven't tried this on Noetic yet but I believe some users managed to install this successfully on Noetic.  

You can check out this snippet to make python-gudev work on 18.04 above. 
https://github.com/chvmp/champ_install/blob/master/install#L97-L114

Hope these help.

pitosalas, 2021-01-19:
Thanks. I dropped down to using the Ubiquity Robotics pi image and that solved my problem. I figured I might be able to solve this one problem but there would be others waiting so I might as well go to a more mainstream setup!

Thank you for a great set of packages!

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2021-01-24:
By the way, you mean both motors, right, not just one?

Its hard to tell whether they are right or wrong because as soon as I do what you suggested and tried to make it go slow 0.2 it immediately goes into the rapid back and forth behavior.

Pito Salas
Faculty, Computer Science
Brandeis University

> On Jan 24, 2021, at 10:32 AM, Przemek Dyszczyk <pdyszczyk@gmail.com> wrote:
> 
> I think you can change port setup - just swap them 
> 
> On Sun 24 Jan 2021, 15:29 Pito Salas, <pitosalas@gmail.com> wrote:
> Thanks! Do I have to change the cables, or can I change the port setup?
> 
> -- Pito
> 
> 
> On Sun, Jan 24, 2021 at 5:54 AM Przemek Dyszczyk <pdyszczyk@gmail.com> wrote:
> Hi,
> It was my mistake, I didn't know that I send it only to you :)
> 
> It looks like your wheels have incorrect polarization and they are not rotating forward - so you have to swap cables that give them power. Try to lower P to 0.2 and rotate them slowly. 
> About PID configuration - I think that we have only that /pid_configure in rqt that you have in the lower right corner. I would try with P=0.2, I=0, D=0 and make sure that motors are rotating correctly
> 
> sob., 23 sty 2021 o 17:53 Przemek Dyszczyk <pdyszczyk@gmail.com> napisał(a):
> Hi Pito,
> That is very common issue there. From the reading I can tell that encoders for your rare wheels are not connected properly. Please, check all the cables for encoders.
> The best way to check if encoders are working and that they are connected in proper way (value is increasing when going forward) is to move wheels by hand. When all 4 of them works you can try to configure PID.
> 
> And it is that issue 100% :)
> Good luck!
> 
> sob., 23 sty 2021 o 16:04 Pito Salas <pitosalas@gmail.com> napisał(a):
> I've built a new robot, and I believe I have all the software installed correctly. I was able to callibrate the IMU, run minimal.launch. While this is a new robot for me I am very familiar wth ROS.
> 
> Whenever the robot receives a command to move, the wheels run very fast and after the first /cmd_vel it does not respond to any additional ones. On the assumption that some of my pinouts were incorrect I reversed the motor direction pins, and I also tried reversing the encoders. But the problem persists. Here is the output from minimal. Those numbers look suspect but I don't know what they mean.
> 
> Once I send the first cmd_vel (either via the teleop script or rostopic pub) I see the following output. And no amount of additional cmd_vels, with speeds set to zero, will stop the spinning. I have to power off.
> 
> Can you help me figure out what is wrong? Thanks!
> 
> [INFO] [1611417374.071803]: Encoder RearRight  : 0
> [INFO] [1611417374.277823]: Encoder FrontLeft  : 679840
> [INFO] [1611417374.281453]: Encoder FrontRight : -681316
> [INFO] [1611417374.285495]: Encoder RearLeft   : 0
> [INFO] [1611417374.289098]: Encoder RearRight  : 0
> [INFO] [1611417374.496340]: Encoder FrontLeft  : 681210
> [INFO] [1611417374.501486]: Encoder FrontRight : -682698
> [INFO] [1611417374.506152]: Encoder RearLeft   : 0
> [INFO] [1611417374.510884]: Encoder RearRight  : 0
> [INFO] [1611417374.714882]: Encoder FrontLeft  : 682579
> [INFO] [1611417374.719173]: Encoder FrontRight : -684080
> [INFO] [1611417374.723595]: Encoder RearLeft   : 0
> [INFO] [1611417374.728066]: Encoder RearRight  : 0
> [INFO] [1611417374.933741]: Encoder FrontLeft  : 683948
> [INFO] [1611417374.938679]: Encoder FrontRight : -685462
> [INFO] [1611417374.942401]: Encoder RearLeft   : 0
> [INFO] [1611417374.946195]: Encoder RearRight  : 0
> [INFO] [1611417375.151942]: Encoder FrontLeft  : 685318
> [INFO] [1611417375.157373]: Encoder FrontRight : -686844
> [INFO] [1611417375.161392]: Encoder RearLeft   : 0
> [INFO] [1611417375.165274]: Encoder RearRight  : 0
> [INFO] [1611417375.371809]: Encoder FrontLeft  : 686687
> [INFO] [1611417375.376137]: Encoder FrontRight : -688228
> [INFO] [1611417375.379986]: Encoder RearLeft   : 0
> [INFO] [1611417375.383901]: Encoder RearRight  : 0
> [INFO] [1611417375.590633]: Encoder FrontLeft  : 688058
> [INFO] [1611417375.594883]: Encoder FrontRight : -689611
> [INFO] [1611417375.598370]: Encoder RearLeft   : 0
> [INFO] [1611417375.602425]: Encoder RearRight  : 0
> [INFO] [1611417375.809028]: Encoder FrontLeft  : 689428
> [INFO] [1611417375.813647]: Encoder FrontRight : -690994
> [INFO] [1611417375.817237]: Encoder RearLeft   : 0
> [INFO] [1611417375.820785]: Encoder RearRight  : 0
> [INFO] [1611417376.028545]: Encoder FrontLeft  : 690798
> [INFO] [1611417376.033642]: Encoder FrontRight : -692376
> [INFO] [1611417376.037465]: Encoder RearLeft   : 0
> [INFO] [1611417376.041365]: Encoder RearRight  : 0
> [INFO] [1611417376.247802]: Encoder FrontLeft  : 692166
> [INFO] [1611417376.252402]: Encoder FrontRight : -693757
> [INFO] [1611417376.256212]: Encoder RearLeft   : 0
> [INFO] [1611417376.260079]: Encoder RearRight  : 0
> [INFO] [1611417376.466706]: Encoder FrontLeft  : 693536
> [INFO] [1611417376.471168]: Encoder FrontRight : -695139
> 
> -- 
> You received this message because you are subscribed to the Google Groups "LINOROBOT" group.
> To unsubscribe from this group and stop receiving emails from it, send an email to linorobot+unsubscribe@googlegroups.com.
> To view this discussion on the web visit https://groups.google.com/d/msgid/linorobot/e8b951a9-8d22-443a-8ddb-ff59790c4c4cn%40googlegroups.com.

IAseven, 2021-02-02:
@pitosalas Can you describe your problem in detail?

pitosalas, 2021-02-02:
I since wrote up a very detailed post on how I troubleshot the problem: https://groups.google.com/g/linorobot/c/egaPyuaEyns

It was some combination of (zero or more) 1) one motor driving reverse, 2) one encoder being backwards and the 3) IMU being mounted in the wrong orientation. In the end I don't know which of those was the culprit but my writeup above took me through a methodical troubleshoot which eventually allowed me to solve it.