linorobot/linorobot #57: Angular velocity bug

IAseven, 2021-01-16:
For 2WD platform, when calculating the angular velocity is half of the actual value.
In the kinematics file, the getvel function is calculated incorrectly, the code is as follows:
` vel.angular_z =  (average_rps_a * wheel_circumference_) / ((wheels_x_distance_ / 2) + (wheels_y_distance_ / 2));`
Here, 4WD and 2WD are calculated separately.

grassjelly, 2021-01-19:
Hi there, the wheels_x_distance_ is set to 0 on 2wd 
`    wheels_x_distance_(base_platform == DIFFERENTIAL_DRIVE ? 0 : wheels_x_distance),
`
https://github.com/linorobot/linorobot/blob/master/teensy/firmware/lib/kinematics/Kinematics.cpp#L35

and rpm3 and rpm4 are 0s as it is assumed to have no encoder readings

IAseven, 2021-01-26:
thanks for you.

closed 2021-01-26 by IAseven