linorobot/linorobot #59: Status of this project?

pitosalas, 2021-01-26:
Hi @grassjelly -- what a wonderful project this is. I am having a great time with it. I was wondering whether it is still actively maintained or whether  you have moved to other things. It would help me understand whether I want to build my stuff around it. Thanks!

pitosalas, 2021-01-31:
I am thinking that the lack of response means that @grassjelly is no longer actively engaged? Is there another official maintainer?

IAseven, 2021-02-01:
@pitosalas As far as I know, the author currently focuses on the quadruped robot project, link: https://github.com/chvmp

Russ76, 2021-02-01:
@pitosalas If you build with nearly the same thing as in these lists, it should still work. There might be some small issues. The Python version 2.7 is not supported by Platformio now, and I'm trying to work around that. I'm also using ROS Melodic. I'd say, Dive In! Juan still does answer issues here but in his time.

pitosalas, 2021-02-01:
Ok. So there's no active maintenance just the occasional response from @grassjelly . I will try a pull request and see what happens!

closed 2021-02-01 by pitosalas