linorobot/linorobot #61: Platformion requires python 3.6 or newer

Russ76, 2021-02-03:
This issue is difficult. To install the Teensy code requires ROS routines (which use Python 2.7) but Platformio won't work with that old version. I made a custom version of Ubuntu 18.04 on a micro SD card (for my Raspberry Pi 4) and removed Python 2.7 (which removed most of ROS Melodic) and then Platformio installed well. But to compile and transfer code to Teensy, it requires many ROS items for the code! I don't find a simple answer.

pitosalas, 2021-02-13:
I had this problem and got this response on the platformio issues list: https://github.com/platformio/platformio-docs/issues/176

Russ76, 2021-02-13:
Thanks for that! I'll take a close look at it and see if it works.

closed 2021-02-13 by Russ76