linorobot/linorobot #68: ModuleNotFoundError: No module named 'glib'

pitosalas, 2021-06-15:
Hi! It looks like you are in the middle of doing the same thing that I am, but two weeks ahead or so. I am trying to use the Linorobot stack on a Raspberry Pi 3+ running Ubuntu 20.02 server. (I know you are not on ubuntu but let's plow on.) I've solved many of the problems caused by the Python 3 requirement but I am a little stuck on the error above:

```
rosrun lino_udev lino_udev.py
Traceback (most recent call last):
  File "/home/ubuntu/linorobot_ws/src/linorobot_src/lino_udev/scripts/lino_udev.py", line 2, in <module>
    import glib
ModuleNotFoundError: No module named 'glib'
```

Do you know what the solution is? Thanks!

NickNair, 2021-08-22:
Have you tired importing it like this ?
```
from gi.repository import GLib as glib
```

sadiq26, 2023-01-29:
What was the solution to this. Facing the same issue.