linorobot/linorobot #69: Install script doesn't complete and gives "Python 3.6 or greater required" error

andrewrgross, 2021-08-19:
During installation on a Jetson Nano running Ubuntu 18.4, the install script gave this error. Collectively, a group of us worked around it by editing the install script to remove lines that seemed to cause the problem that weren't relevant to our project.

andrewrgross, 2021-08-19:
Here is our edited install file:

```
#!/usr/bin/env bash

set -e

source /opt/ros/$(dir /opt/ros)/setup.bash

sudo cp files/49-teensy.rules /etc/udev/rules.d/

ROSDISTRO="$(rosversion -d)"
BASE=$1
SENSOR=$2
ARCH="$(uname -m)"
echo $ARCH
echo "
______ _____________   _________ ________ _______ ________ _______ ________
___  / ____  _/___  | / /__  __ \___  __ \__  __ \___  __ )__  __ \___  __/
__  /   __  /  __   |/ / _  / / /__  /_/ /_  / / /__  __  |_  / / /__  /   
_  /_____/ /   _  /|  /  / /_/ / _  _, _/ / /_/ / _  /_/ / / /_/ / _  /    
/_____//___/   /_/ |_/   \____/  /_/ |_|  \____/  /_____/  \____/  /_/     
                    
                            http://linorobot.org                                                                          
"
if [ "$3" != "test" ]
    then
        if [ "$*" == "" ]
            then
                echo "No arguments provided"
                echo
                echo "Example: $ ./install.sh 2wd xv11"
                echo
                exit 1
                
        elif [[ "$1" != "2wd" && "$1" != "4wd" && "$1" != "mecanum" && "$1" != "ackermann" ]]
            then
                echo "Invalid linorobot base: $1"
                echo
                echo "Valid Options:"
                echo "2wd"
                echo "4wd"
                echo "ackermann"
                echo "mecanum"
                echo
                exit 1

        elif [[ "$2" != "xv11" && "$2" != "rplidar" && "$2" != "ydlidar" && "$2" != "hokuyo" && "$2" != "kinect" && "$2" != "realsense" ]]
            then
                echo "Invalid linorobot sensor: $2"
                echo
                echo "Valid Options:"
                echo "hokuyo"
                echo "kinect"
                echo "lms1xx"
                echo "realsense"
                echo "rplidar"
                echo "xv11"
                echo "ydlidar"
                echo
                exit 1
        
        elif [[ "$ARCH" != "x86_64" && "$2" == "realsense" ]]
            then
                echo "Intel Realsense R200 is not supported in $ARCH architecture."
                exit 1

        fi

        echo
[... 136 lines cut ...]
catkin_make

cd $HOME/linorobot_ws/src
git clone https://github.com/linorobot/linorobot.git
git clone https://github.com/linorobot/imu_calib.git
git clone https://github.com/linorobot/lino_pid.git
git clone https://github.com/linorobot/lino_udev.git
git clone https://github.com/linorobot/lino_msgs.git

cd $HOME/linorobot_ws/src/linorobot
TRAVIS_BRANCH="echo $TRAVIS_BRANCH"
if [ "$TRAVIS_BRANCH" = "devel" ]; then git checkout devel; fi

echo "source $HOME/linorobot_ws/devel/setup.bash" >> $HOME/.bashrc
echo "export LINOLIDAR=$SENSOR" >> $HOME/.bashrc
echo "export LINOBASE=$BASE" >> $HOME/.bashrc
source $HOME/.bashrc

cd $HOME/linorobot_ws
catkin_make

echo
echo "INSTALLATION DONE!"
echo
```

closed 2021-08-19 by andrewrgross

knnurl, 2022-08-26:
You guys are amazing, thank you.